nakhonthai/ESP32APRS_Audio #5: Feature Request: daughter card or optional space for an sa818

derekmurawsky, 2023-11-01:
This is a great project. I think it would be awesome to extend the board or create a daughter card for it which could take an SA818 to make an all-in-one APRS digipeter/igate. I could also see a similar daughter card for a LoRa radio, which seems to be gaining popularity in amateur radio circles. 
I'd love to help with something like this, but my embedded hardware/software skills are limited. Happy to support with testing, though.

nakhonthai, 2023-11-03:
I'm thinking of creating a new project ESP32APRS for All-in-One.

derekmurawsky, 2023-11-04:
I'd be happy to provide input and testing... Even code, though I'm not the best there. I believe there is a need for a true, simple, open source hardware APRS tracker/digi/igate and I suspect the esp32 would be able to handle it all.

erstec, 2023-11-04:
> I'd be happy to provide input and testing... Even code, though I'm not the best there. I believe there is a need for a true, simple, open source hardware APRS tracker/digi/igate and I suspect the esp32 would be able to handle it all.

Ir handles already, several devices are constantly online with ESP32 for last year+ https://github.com/erstec/APRS-ESP
More, it have official APRS TOCALL

nakhonthai, 2023-11-13:
I'm update in the Version 1.0 ,you can add and mod GPIO of RF Modules.

closed 2023-11-13 by nakhonthai

derekmurawsky, 2023-11-13:
Awesome!