nakhonthai/ESP32APRS_Audio #6: No Access point connexion

f4goh, 2023-12-25:
Hello,
i load last release 11 with oled.
I see ESP32Igate after factory reset
try to connect with aprsthnetwork pass.
but ESP32Igate disconnect any client.
and no ESP32Igate wifi network (esp32 wifi down ?)
how to proceed ?
thanks

nakhonthai, 2023-12-26:
Use wifi client mode.
Share wifi from router or mobile phone.
SSID: APRSTH
Pass: aprsthnetwork

Example clipt on youtube https://www.youtube.com/watch?v=e9tvUChsF4U

f4goh, 2023-12-26:
Ok 6.2 config works but not 6.1. you confirm there is a problem. Thanks.
6.2 Default config then go to share or set router WiFi SSID: APRSTH PASS: aprsthnetwork and use web browser by IP from router/share smartphone default web auth USER: admin PASS: admin

closed 2023-12-26 by f4goh