nakhonthai/ESP32APRS_Audio #12: 1.11D tx ptt problem

TA1FM, 2024-01-21:
I was using v0.9, I installed v1.11D, the TX LED does not light up and it does not switch the radio to TX. There was no such problem in v0.9.

cabralcr, 2024-01-24:
Here the ptt is working normally

nakhonthai, 2024-01-26:
Check in MOD Tab of GPIO and active pinout.

TA1FM, 2024-01-26:
I made the esp32 igate circuit in v08, I connected it to an external handheld radio, I did not use a module like SA668, tx led and ptt were working fine, when I installed v1.1D on the same system, tx led and ptt did not work.
Even though I made the PTT GPIO settings in the MOD tab in the interface, it does not work. I tried many of the settings in the RADIO tab, but it still did not work. I could not figure out why. When I go back to the old v08 or v09 software, it works without any problems.
I follow your work with pleasure, thank you for your interest.
TA1FM
[image]

nakhonthai, 2024-01-30:
If SQL pin signal not used, it to  fix  SQL  GPIO = -1

TA1FM, 2024-01-30:
I'm sorry to take your valuable time, but when I set SQL GPIO to "-1", esp32 keeps resetting during TX.
https://photos.app.goo.gl/8rRFmnzmNKU61nzv5

nakhonthai, 2024-01-31:
I've fixed this in the version 1.2b, please update firmware again.

TA1FM, 2024-01-31:
I installed V1.2b, but during TX, ESP resets again. If I turn off the INET2RF switch, there is no problem as only igate.

TA1FM, 2024-03-04:
There is no problem in the software without an OLED screen, but in the software with an OLED screen, the problem was solved when I added 2 pull-up 4k7 resistors to the data paths of the screen.

closed 2024-03-04 by TA1FM