nakhonthai/ESP32APRS_Audio #17: Feature Request: transmit an emergency beacon

nicolassolop, 2024-02-01:
Hello, let me tell you that I think your project is great and very useful.

With a group of friends we go out to the mountains frequently and it would be ideal to be able to carry an APRS tracker.

Something important for us would be to have the possibility of transmitting an emergency beacon in case of having a problem.

We thought it would be an option to be able to press a button for a few seconds and have this trigger an emergency beacon.

What do you think about the idea?

nakhonthai, 2024-02-01:
Thank you for the advice.
I already have this idea. It is sending urgent information for emergencies. or notify various statuses in real time.

nicolassolop, 2024-02-07:
Yes, that would be very usefull for a lot of people. Would be great to have this in future releases