nakhonthai/ESP32APRS_Audio #19: Bug: V1.2b - IGate -> RF causes ESP32 to reset.

M0XZS, 2024-02-02:
Since moving from V1.1 to 1.2, I have noticed that the ESP starts to reset when the INET to RF is turned on in the IGGate. 

I am using an external radio (with RF shielding) - nothing changed other than the firmware, and it looks like the transmitter does into an extended transmission with no APRS modulation, and then the ESP32 resets.

I did a Diff on the V1.1 and V1.4b changes, and I can not see any reason this would happen.

M0XZS, 2024-02-02:
I put the debugging on and found this just before the reset:

[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[W][esp32-hal-i2c.c:1419] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=1
[D][esp32-hal-i2c.c:1427] i2cCheckLineState(): Recovered after 1 Cycles
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[W][esp32-hal-i2c.c:1419] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=1
[D][esp32-hal-i2c.c:1427] i2cCheckLineState(): Recovered after 1 Cycles
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
E (176382) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (176382) task_wdt:  - IDLE0 (CPU 0)
E (176382) task_wdt: Tasks currently running:
E (176382) task_wdt: CPU 0: taskAPRS
E (176382) task_wdt: CPU 1: IDLE1
E (176382) task_wdt: Aborting.
abort() was called at PC 0x4018d32c on core 0

M0XZS, 2024-02-02:
Returning Task 1 to priority 1 seems to fix the problem:

    // Task 1
    xTaskCreatePinnedToCore(
        taskAPRS,        /* Function to implement the task */
        "taskAPRS",      /* Name of the task */
        8192,            /* Stack size in words */
        NULL,            /* Task input parameter */
        1,               /* Priority of the task */
        &taskAPRSHandle, /* Task handle. */
        0);              /* Core where the task should run */

M0XZS, 2024-02-02:
This did improve it but did not fix it.

M0XZS, 2024-02-05:
This seems to be the problem, but not sure why yet:

[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit

and it gets into a loop.

TA1FM, 2024-02-29:
Hello, can you explain how you solved the problem in more detail, we would like to benefit from it. Can you also add the corrected fiemware.bin file, thank you.
TA1FM