nakhonthai/ESP32APRS_Audio #54: ESP32C3 release firmware has problems

YuYanDev, 2025-09-08:
The standard firmware package doesn't include the ESP32C3 firmware. So I chose the 8MB firmware package (it seems there's no 8MB version for the ESP32C3?). 

After the flash was complete (the flashing tool indicated success), I didn't find any APRS Wi-Fi hotspots. 

To this end, I tried checking the serial port information, which was printed as follows:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
Invalid chip id. Expected 5 read 9. Bootloader for wrong chip?
ets_main.c 333 
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
Invalid chip id. Expected 5 read 9. Bootloader for wrong chip?
ets_main.c 333 
```

There seems to be a problem with the released firmware. Maybe the bootloader is not universal? 
Hope you can solve this problem. Thank you.

YuYanDev, 2025-09-08:
After I successfully flashed using Platformio, I could access the wifi and enter the configuration backend. However, there was a problem with the firmware and the configuration could not be updated successfully. The interface reported an error.

nakhonthai, 2025-09-08:
Firmware ESP32-C3 at https://github.com/nakhonthai/ESP32APRS_Audio/releases/download/V1.5/ESP32C3_V15b.zip
Check address for bootloader,boot_app,partition and firmware

YuYanDev, 2025-09-08:
It works ! thank you

closed 2025-09-08 by YuYanDev